Blue-Smoke-Electronics/Ember #158: removable battery and capacitor instead

jerabaul29, 2022-11-24:
I would like to avoid having more lithium batteries home - these batteries, when getting discharged low enough, are a fire hazard. Would it be possible then to:

- make the battery easily removable, with a battery holder or something like this, and selling the unit optionally with or without battery
- add (a couple of, depending on what voltage is needed) supercapacitor(s) or lithium ion capacitor(s), so that it remains possible to use the unit with some power bursts even without the battery inserted (note: supercapacitors may be best to use, as it is fine to discharge them down to 0; if using lithium ion capacitors, these are like small batteries on some aspects and cannot be drained to 0V, so monitoring of their voltage would be necessary)

simenhs, 2022-11-24:
To remove the battery, you need to screw out 4 screws and disconnect one JST connector, so it is possible to remove/replace the battery. 

I really like your idea of using super capacitors. It would be nice to not have to deal with batteries, and all of their drawbacks. I have never really played around with super caps, do you have any experience? Will it be possible to fit them inside the enclosure? I guess I will have to settle for a smaller capacity, but a big reason that I am not using a smaller battery is because of voltage drop over the internal series resistance, and that would not be a problem with supercap.

jerabaul29, 2022-11-24:
I do have played around with super capacitors (supercaps) quite a bit. I have not played with lithium ion capacitors (LICs) too much. I think anyways that, for your kind of application (taking away spikes), supercaps are much easier to use than LICs because it is not an issue if their voltage goes to 0 as long as the voltage impose does not becomes negative.

My 2 cents on supercaps:

- they can be really powerful; I have used some 60F supercaps in parallel to take off power spikes in a linear actuator - something like 10 supercaps in series to get the 25V voltage I needed, it was super effective
- I have not had issues in handling them
- they are not sooo big (you can find some quite small supercaps with 10 or 20 or 30 F capacity; this is really a lot, can easily power electronics in case there is a small spike of a few amps for a few seconds: dU / dt = I / C, so at 10 A and 15F, you loose 0.6V per second!
- they are a bit slower than typical electrolytic caps; so if you want something really neat, may be a good idea to have (for a 5V supply) 2 supercaps in series, in parallel of 2 electrolythic caps in series (possibly in parallel to a few ceramic caps if you want to take out real fast voltage fluctuations).

simenhs, 2022-11-24:
Just some calculations: 
When at 100 % output i draw 
15V*1A =15W and have around 5W of internal losses -> 20W total power consumption from battery. 
I want this last be able to deliver this for atleast 1 min 
20W * 60s = 1200 J 
min V for device to work - 2.5V  supercap max voltage 2.7V 
5000 F -> 5000/2*2.7*2.7 = 1350 J    5000/2*2.5*2.5 = 1200 J   13500-12000 = 1500J 
5000 F is around 150 usd 

LIC caps 
can go up to 3.8V which means i can use a 400 F cap. price: 20 usd (2x220F)

With two super caps in series should be good, I can charge them to 5V and use 75% of their power before they drop below 2.5V. Two 60 F should give me around 18 seconds of maximum power draw.  

Will have to test it out, ordering some supercaps and LICs now.

jerabaul29, 2022-11-24:
A few notes / bainstorming / making myself the devils advocate :) :

- the LIC can only be discharged down to 2.5V or something like this (need to check), so some control electronics are needed to make sure the LIC is not over discharged and damaged if you decide to use LICs

- if we assume users that actively choose to have no battery then they always use a USB input, is it really necessary to hold for 1 min? Maybe just taking away the power spikes (let us say 10s) is enough? Then 2 supercaps may be enough?

simenhs, 2022-11-24:
I do already have the protection in place, batteries can not be discharged below 2.5V as well. So that should not be a problem. Production on the other and will be easier if i do not have to worry about shorting the terminals and draining the capacitor. 

18 seconds might be enough in many use cases, I agree. Drawing 15W is not going to happen very often ether( the load need to be 15 ohm exactly). I guess the most probable case when a larger battery will be nice is when you are not connected to 5V, like when you are outside. You could bring a power bank... 

How did you deal with balancing when you put many supercaps in series? If one of the capacitors gets charged more than the other the chance of one of them being damaged is large if i am splitting 5V over two capacitors(2.8V+2.2V = blue smoke).

jerabaul29, 2022-11-24:
Sounds good.

I have applied passive balancing using 1 resistor of well chosen value in parallel of each capacitor. There are other methods though for example using an op amp. See for example for explanations:

https://www.we-online.com/catalog/media/o671684v410%20ANP090a_EN.pdf
https://www.maximintegrated.com/en/design/technical-documents/app-notes/7/7552.html
https://www.electronicdesign.com/power-management/whitepaper/21134743/balancing-supercapacitor-stack-voltages

Since you have an external power source with the USB cable and you do not need to be low power, you can take some large safety margin (ie have lower resistance resistors than what is strictly needed so that more balancing current than strictly necessary can flow).

simenhs, 2022-11-26:
Thanks, a joy full read. 
I think resistors may be good enough when the USB cable is plugged in, as you say, I am not too worried about wasting power then. But I will have to find a way to disconnect the resistors when the charger is disconnected. I do not want the capacitors to always be drained when you plug in and turn on the device. I guess two small transistors easily can fix that.

simenhs, 2022-12-02:
[image]

simenhs, 2022-12-02:
[image]

jerabaul29, 2022-12-02:
Looks excellent!

jerabaul29, 2022-12-02:
So you put diodes to avoid revert polarity and overvoltage by choosing the forward voltage suitably I guess?  Looks good :)

simenhs, 2022-12-02:
Tested with two 100F super capasitors in series today. It worked like well. I got around 30 seconds off max output draw, and around 1 hour of idle time. Using resistors as ballansing do not work, it is way to show with reasonable losses. I will test out zener diodes next. Using the battery charger to charge the supercaps works, kinda, but max 4.1V and slow charge below 3V is not ideal. I have one opamp extra in the quad pac I am not currently using, so I am thinking of using that and creating a constant current charger for the supercaps.

simenhs, 2022-12-02:
yes, the diodes is suppose to be 2.7V zener diodes, to protect the individual cells from overvoltage. I guess the also protect from reverse current, but I really hope that that will never happen.

simenhs, 2022-12-02:
Two 100 F supercaps has almost the same size as the batteries, so it will not impact the size of the device to much.

simenhs, 2022-12-05:
https://www.eevblog.com/forum/projects/battery-or-supercapasitor-in-portable-bench-power-supply/

simenhs, 2023-01-02:
Added holder for 18650 li-ion battery on rev 10.

closed 2023-01-02 by simenhs